Hyperloop-UPV/ST-LIB #5: Problema de eficiencia en el modulo de la FLASH

PabloGonzalezSegarra, 2022-10-19:
El problema que tenemos es el siguiente, la memoria flash del los H723x solo se puede escribir por sectores, pues para poder escribir primero tienes que borrar la parte en la que quieres escribir pero este borrado solo se puede hacer por sectores, que en el H723 es de 128KB.
[image]
Esto conlleva que si queremos escribir por ejemplo 1KB de datos primero tenemos que leer los 128KB de datos, después borrarlos todos, una operación que tarda bastante,  seguidamente escribimos nuestros datos y rellenemos lo que no hemos escrito con los datos que hemos leído antes. Esta operación completa puede llegar a afectar a un sistema en tiempo real como el nuestro, habría que probar experimentalmente hasta que punto es crítico.
Si solo tenemos que escribir muy pocas veces y en momentos no críticos esto no debería suponer un problema, si si que nos supondría un problema hay dos soluciones.
La primera es usar una memoria EEPROM externa con sectores mucho más pequeños, la segunda es implementar soluciones por software, la primera es gestionar el borrado y la escritura de la flash por interrupciones, esto alargaría el tiempo de desarrollo de este modulo significativamente pues habría que implementar una cola de peticiones de escritura y tener en cuenta la fase en la que nos encontramos de cada operación de escritura. Esta opción no reduciría el tiempo de escritura pero liberaría el microcontrolador para hacer otras operaciones mientras se esta escribiendo en la flash. La segunda opción por software si que haría más rápida la escritura y se trata de emular una memoria EEPROM a través de un journal, STM32 tiene algunas implementaciones pero no son compatibles con el H723 por esto habría que implementar nosotros mismos nuestra versión y por la información que he recabado es muy poco recomendable.

StefanCostea, 2022-10-20:
Ninguna placa va a escribir de normal nada en la flash jamás, no queremos que una placa se pueda poner a hacerlo mientras corremos y se corrompa. La EEPROM ni de coña chaval xD. Para empezar el planteamiento que tenemos de momento esque las placas ni escriban en la flash solo lean, sin embargo por añadir funcionalidad dar la posibilidad de poder escribir en ella es util, te pongo un ejemplo: 
Están probando levitación y cambiando parámetros como simios, el que este al cargo de subir código a la placa no está haciendo nada más, ya que es el unico que sabe cómo va eso y no hace otra cosa durante horas que no sea cambiar un array y darle al play. Podríamos guardar la kp ki y kd en la flash y que la placa pueda recibir una orden con los nuevos parámetros y que los sobreescriba, de esta manera no hace falta reprogramar la placa.

StefanCostea, 2022-10-20:
Entonces yo implementaría algo que solo te ofrezca esas dos funcionalidades leer y escribir y que estén preparadas para hacerlo de manera muy puntual pero intenta simplificarlo, podríamos añadir a ADE la información sobre donde se almacenan las cosas de la placa en la FLASH así ya no tendrías que complicarte tanto la vida con lo que hablamos ayer pero bueno esto ya es pensar.

PabloGonzalezSegarra, 2022-10-20:
Vale, perfecto. Lo voy a hacer simplificado y lo mejor es que la ADE guarde donde están las cosas y luego se generen defines con la dirección.

closed 2022-10-20 by PabloGonzalezSegarra