Hyperloop-UPV/ST-LIB #7: ¿Como vamos ha hacer el manejo de las excepciones?

PabloGonzalezSegarra, 2022-10-20:
Por ejemplo en el siguiente código se puede llegar a producir una excepción a la hora de escribir en la flash.
[image]
¿Qué estrategia vamos a seguir para manejar estas excepciones? ¿lo dejamos como TODO mientras no tengamos los warnings propios implementados?

alejandrogonzalvo, 2022-10-20:
Como mínimo debería enviarse algún tipo de advertencia en forma de paquete a la gui o a la terminal de debug o lo que estemos utilizando y luego @StefanCostea quiere implementar warnings este año, estaria muy chulo que cuando usas una función que devuelve un HAL_STATE y no haces lógica condicional en función del return te envíe un warning

StefanCostea, 2022-10-20:
Para eso estan los warnings, lo comenté ya ayer los warnings van a ser mensajes que lleguen a la gui como strings y sirvan para debuggear cosas de estas. Al usar ETH ya no hay problema de colisión de ids con algunas cosas ya que el id forma parte del mensaje asi que sera un mensaje con determinada id con un string dentro.
Ejemplos de Warning:

WARNING: Control loop of LPU is running below 5kHz
WARNING: Trying to assign pin twice in board with id 21

StefanCostea, 2022-10-20:
Lo que dice @alejandrogonzalvo es imposible ya que si no recoges el HAL_State se borra, la solucion seria recogerlo y poner el warning en el error handler(no la mejor), o no preocuparse porque como la halal envuelve toda la hal nunca habrá que hacer un hal start explicito estará en vuelto en alguna función, por ejemplo: Pin::Start() Communication::init() y en los determinados init o funciones que rodeen estas cosas hacer el manejo del HAL_State y mandar un warning

StefanCostea, 2022-10-20:
Hasta que tengamos los Warnings me parece bien lo que dice @PabloGonzalezSegarra de hacer una todo-list con sitios donde habrá que añadirlos

PabloGonzalezSegarra, 2022-10-20:
Vale, entonces por ahora vamos dejando un **TODO** en los sitios donde se debería emitir un **warning**.

closed 2022-10-20 by PabloGonzalezSegarra