Hyperloop-UPV/ST-LIB #9: Return value de funciones register cuando no se ha podido registrar el pin correctamente

alejandrogonzalvo, 2022-10-20:
Las funciones register de las ADC, PWM, Digital... devuelven un uint8_t. Hay casos en los que el pin no se ha podido registrar, por ejemplo cuando se intenta registrar en un pin que no está en el mapa de pines del servicio, (que corresponderá con el IOC o las runes). En estos casos debería emitirse un warning seguro, pero hay que ver que devuelven entonces. No pueden devolver -1 porque es un uint, yo propongo que devuelvan 0, y el valor 0 se elimine de todos los IDmanager. De esta forma se reserva ese número y las funciones lógicas superiores saben si el pin no se ha registrado correctamente solo con comprobar si devuelve cero.

alejandrogonzalvo, 2022-10-20:
La otra opción es cambiar el return a un signed int y que devuelva -1

alejandrogonzalvo, 2022-10-20:
Una tercera opción seria implementar los optionals, pongo un extracto de codigo de como quedaria con optionals.

`optional<uint8_t> ADC::register_adc(Pin pin) {
	if (pinTimerMap.find(pin) != pinTimerMap.end()) {
		Pin::Register(pin, ALTERNATIVE);
		uint8_t id = IDmanager.front();
		ADC::serviceIDs[id] = pin;
		ADC::IDmanager.pop_front();
		return id;
	}

	return nullopt; // WARNING TODO
}`

StefanCostea, 2022-10-20:
Yo lo que mejor veo es el tema de los optionals

StefanCostea, 2022-10-20:
Justo en digitaloutput e input no lo he puesto porque todos los pines tienen esa opción pero en otros servicios yo haria lo del optional y au

StefanCostea, 2022-10-20:
Que te devuelva una id 0 y reservarla es literalmente lo mismo que un optional ya que tendrías que seguir implementando el checkeo fuera y es mas limpio, claro y no pierdes IDs usando optionals