Hyperloop-UPV/ST-LIB #11: La instrucción unordered_map<Pin, nonimportant>.find(Pin) nunca devuelve true.

alejandrogonzalvo, 2022-10-22:
En las funciones register se debe incluir una protección para cuando se intente registrar en un pin un periférico que no le corresponde. Para ello creo un mapa que indica todos los Pines disponibles, y si no lo encuentra devuelve nullopt. Para buscar si el pin a registrar está en el mapa, se utiliza la función find. En principio pensé que utilizaria el operador == implementado por mí en la clase Pin, pero he puesto que siempre devuelva true y sigue sin encontrar el pin. En cambio hacer un bucle for que itere todas las llaves del mapa y la compare con el pin a registrar si funciona. De momento, voy a hacerlo así para que funcione estaria mejor hacer funcionar el find.

Metodo find:

StefanCostea, 2022-10-22:
Es porque la clase pin no implementa el operador == de manera que cuando se comparan el parámetro que le pasas a map.find()
y el objeto almacenado en el iterador solo comprueba si ambos comparten referencia. Solucionarlo seria tan fácil como implementar el siguiente operador en la clase pin:
    bool operator==(const Pin &rhs) const
    { 
        return (this->pin == rhs.pin && this->port == rhs.port);
    }

alejandrogonzalvo, 2022-10-22:
bool operator== (const Pin &other) const {
		return (pin == other.pin && port == other.port);
	}
Esta es la función que decia que he implementado, y ademas he probado a cambiarla por return true, y sigue sin funcionar.

StefanCostea, 2022-10-22:
Pero no en el KeyEqual del mapa, sino en la propia clase Pin

StefanCostea, 2022-10-22:
de hecho si se implementa en la propia clase pin no hace falta implementar un KeyEqual

alejandrogonzalvo, 2022-10-22:
Eso es lo que he hecho, no funciona

alejandrogonzalvo, 2022-10-22:
```
unordered_map<Pin, TimerChannel, PWM::KeyHash> PWM::pinTimerMap = {
		{PE9, {&htim1, 1}}
};
```

StefanCostea, 2022-10-22:
Enseñame tu KeyHash

StefanCostea, 2022-10-22:
Prueba a ver si el problema es del operador ==, haz una comparación entre 2 pines iguales y a ver que pasa

StefanCostea, 2022-10-22:
@alejandrogonzalvo has hecho el operador == publico?

alejandrogonzalvo, 2022-10-22:
KeyHash:
```
	struct KeyHash {
		size_t operator()(const Pin k) const {
			return hash<uint8_t>()(k.pin) ^
			(hash<uint8_t>()(*reinterpret_cast<uint64_t*>(k.port)) << 1);
		}
	};
```

No, la función del operador es la que he pasado no lleva ni static ni public ni nada delante

closed 2022-10-23 by StefanCostea