Hyperloop-UPV/ST-LIB #14: Modularizar Mapas PIN -> PERIFERICO

alejandrogonzalvo, 2022-10-25:
Actualmente los servicios que utilizan pines tienen un mapa en el cpp con los PINES disponibles que se crea a partir del ioc. Creo que sería mejor meter todos esos mapas en un archivo a parte y utilizar un extern en el cpp en cuestion:

pros:
- cpp más limpio.
- Modularizar configuraciones. (si se quiere otra configuracion, se crea otro archivo a parte, no se edita el servicio directamente).
- Seria mucho más fácil implementar un generador de un archivo runes.hpp a partir de un IOC.

contras:
- no veo

posibles features:
- typedef para los mapas, mucho más limpio leer PinADCmap, que map<Pin, ADCchannel>.
- Ponedme todas las abstracciones y cosas chulas que le pondríais al runes.hpp porfa.

PabloGonzalezSegarra, 2022-10-29:
Me parece una muy buena idea. Este lunes lo hablamos.

alejandrogonzalvo, 2022-11-22:
Esta hecho y en pull request

closed 2022-11-22 by alejandrogonzalvo