Hyperloop-UPV/ST-LIB #18: Add README asap

alejandrogonzalvo, 2022-10-27:
As the scope of this project is to be open source in the near future, we should probably start making some kind of documentation for everyone to read it. Also, the Flash service is getting a bit bloated with comments regarding correct procedures, so we should probably put all that information into a documentation subsection inside the README, and then just keep a comment in the file that says:
_WARNING: READ DOCUMENTATION BEFORE USING FLASH SERVICE._

PabloGonzalezSegarra, 2022-10-27:
It is true that the Flash service requires a little extra knowledge to use it and that it should be a little documented. But I don't think doing it in a README is the right way.

Two things can be done:

-  The first one and simplest is to expand the "Problemas y soluciones de usar la Flash" issue and add what is in the code comment inside the text of the issue.

- The second one that would be better would be to use the github wiki, that is, create a page in the wiki of our repo and add all the necessary information there.

The good thing about the second option is that all the classes can be documented there and thus we manage to have all the technical documentation as well as precautions and instructions for use in a single place and in an organized way.

PabloGonzalezSegarra, 2022-10-29:
Cierro esta issue porque no lo vamos ha hacer en un readme, vamos a probar a documentar con doxygen,

closed 2022-10-29 by PabloGonzalezSegarra