Hyperloop-UPV/ST-LIB #22: How to link an external project with ST-LIB using linked folders.

alejandrogonzalvo, 2022-10-28:
### How to link an external project with ST-LIB using linked folders.
Linking projects with STM32 IDE can mess with relative includes pretty easy. Here is the solution I Found using linked folders.

A linked folder is basically a virtual folder that points to the route of a real one, kinda like pointers.
To create one, we go to the project explorer and click New->Folder. Then we configure it like this:
[image]

Now, we have our folder but we probably can't see it. This is because the project explorer automatically hides certain files. Go into filters and deactivate all options, except .* files:
[image]
[image]

Now you should be able to see your linked folder:
[image]

Next step, go to Project -> Properties and configure your include paths like this:
[image]
[image]
[image]

### Common linking errors
[image]
You haven't activated those ADC and TIM periferal in the ioc. So CUBEMX code generator hasn't generated the appropiate code for those classes. Just enable a timer and an ADC and put a channel to use. Then generate. Remember to change your file to main.c if you are working on a cpp project!

[image]
You are probably trying to compile main.c. Just change it to main.cpp.

[image]
You probably have the ST-LIB project closed. Both the project and the library have to be open for this setup to work.

PabloGonzalezSegarra, 2022-10-28:
No veo la necesidad de la carpeta virtual. Con añadir las siguientes carpetas en el proyecto ya funciona y queda más limpio.
[image]

PabloGonzalezSegarra, 2022-10-28:
El error en el que explicas que hay que añadir uno de cada periférico para que funcione se arregla descomentando en el proyecto que vamos a hacer la hal de todos los periféricos que vamos a tener en la ST-LIB. Como el proyecto base será el mismo esto tendremos que hacerlo solo una vez.

[image]
Es el archivo stm32h7xx_hal_conf.h

alejandrogonzalvo, 2022-10-29:
> No veo la necesidad de la carpeta virtual. Con añadir las siguientes carpetas en el proyecto ya funciona y queda más limpio. [image]

Importante añadirlas tambien en el apartado assembly

closed 2022-10-29 by PabloGonzalezSegarra