Hyperloop-UPV/ST-LIB #41: LookupSensor necesita saber la resolucion para el bit shift

ricardochust, 2022-11-16:
El read de la clase de LookupSensor se basa en obtener el valor binario del pin en el instante actual, meterlo como índice de la LookupTable y devolver lo que haya dentro de ese índice.

Dependiendo del tamaño de la LookupTable y de la resolución del periférico, _deberá ser necesario un bit shift_ para ajustar el valor obtenido en el pin a el tamaño de índices de la tabla. El único problema es saber cual es el rango del valor obtenido en el pin, si trabaja con 12, 11, 8 o cuantos bits. 

¿Va a ser siempre un valor fijo para todos los periféricos que usen la LookupTable, o hago que reciba como parámetro del constructor un valor que sean los bits de resolución de este? Por ahora he asumido que siempre tienen resolución 12 para que se vea como funciona el concepto del código en el ultimo commit a esta clase. 

Y ahora que estamos con el tema, ¿no se tiene pensado que reciba una LookupTable mas grande que la resolución del periférico verdad? No tendría ninguna funcionalidad hacerlo, así que lo he implementado asumiendo que la LookupTable que se le dará será mas pequeña que el rango de valores del periférico, pero si hiciese falta podría hacer que hubiese caso mas grande y mas pequeña (me imagino que los bit shift negativos como que no van)

PabloGonzalezSegarra, 2022-11-16:
Hasta donde yo se va a ser fijo, cuando se conecte Estefan te lo dirá.

StefanCostea, 2022-11-16:
No no lo necesita, el servicio adc tiene metodos que a ti te devuelven el voltaje ya convertido en unidad de coma flotante lo unico que debes hacer es la operación inversa para obtener el valor bruto con la resolución que tu desees.

ricardochust, 2022-11-16:
vale, pues por ahora lo dejo como un define sensor_range 16 y tirando.

closed 2022-11-16 by ricardochust