Hyperloop-UPV/ST-LIB #49: Change Encoder reset to put CNT on 7FFF FFFF instead of 0

ricardochust, 2022-11-23:
The encoder needs to start with a value higher than 0 so if it goes backwards it doesn t suffer underflow.

That on itself is coded on the .ioc configuration, but the Encoder class has a method .reset() that, as the name says, resets the encoder. One of the things that the reset does is change the CNT to the base value. Now it does reset it to 0, which was until now the base value, but as discussed we cannot start at 0 as it has a risk of underflow.

We need to set a base value that we will use on every encoder on the .ioc configuration, and i will suggest 2 147 483 647, or 0x7FFF FFFF, which is half of the maximum, so we balance the risks of underflow and overflow (would fall on 53 687 km without reset with a encoder of 0.025 mm tape). We need to use the same shifted value on the ioc configuration, the encoder::reset method and the encoderSensor::read method so the calculations are consistent. 

If anyone prefers another value just comment it, otherwise i will assume we are using 0x7FFF FFFF

PabloGonzalezSegarra, 2022-11-24:
Ok, as soon as I can I'll get to work on it.

StefanCostea, 2022-11-28:
Is this Issue still open?

StefanCostea, 2022-11-28:
@ricardochust

PabloGonzalezSegarra, 2022-11-28:
I haven't had time to fix it yet. But I hope to do it tonight.

closed 2022-11-30 by PabloGonzalezSegarra