Hyperloop-UPV/ST-LIB #62: Id manager rework

PabloGonzalezSegarra, 2022-11-28:
## Problem
Our id manager was made at the time to move forward and be able to program all the services without thinking much about it. This was fine at first but there has come a time where it doesn't make sense. When the ST-LIB is finished, I see a need for a system rework.

Crime evidence:
[image]
Yes, it hurts to watch.

Apart from being ugly and surely a bad practice, this must be remembered that we are programming a microcontroller with limited RAM memory. Doing a quick calculation of the id lists with the services that are currently implemented in HALAL, we are occupying: 10 services x 256 data / list x (1 byte/data + 4 byte/pointer) = 12.8KB of RAM memory.

## Solution

Obviously this cannot continue at this rate, so I propose the following change. We create a singleton class that is the id manager. This class contains a linked list with the available created ids and a variable that stores the maximum created id. This class will be the same for all services. The internal operation would be very simple, if there are no ids available in the list, one is created that has the value of _last id assigned_ + 1. When a service is left, that id is added to the list as available.
This option is much more optimal in every way, as well as making the use of ids more intuitive.

StefanCostea, 2022-11-28:
I agree with you that that is a lot of RAM wasted, though I don't undertand the solution you propose

alejandrogonzalvo, 2022-11-28:
Let's talk it on wednesday and add it to the trello

PabloGonzalezSegarra, 2022-11-28:
Ok, I'll bring it up at the meeting.

PabloGonzalezSegarra, 2022-12-01:
A first test version will be implemented and if you like it will be applied to all classes.

closed 2022-12-01 by PabloGonzalezSegarra