Hyperloop-UPV/ST-LIB #134: DigitalInput XOR ADC

ricardochust, 2023-01-11:
Parece que hay un problema con el ADC::start(), cuando se usa parece pisar una parte de la inicializacion del Pin::start(), lo cual provoca que el DigitalInput deje de funcionar. 

Aqui el ejemplo de la prueba, un codigo muy simple:

```
/* USER CODE BEGIN 1 */
Pin& myPin = PA3;
Pin& myPin2 = PC0;
DigitalSensor mySensor = DigitalSensor(myPin, myValue);
LinearSensor mySensor2 = LinearSensor(myPin2, 1, 3, myValue2);

 /* Initialize all configured peripherals */
  Pin::start();
  MX_DMA_Init();
  //ADC::start();
  MX_LPTIM1_Init();
  MX_LPTIM2_Init();
  MX_LPTIM3_Init();

/* USER CODE BEGIN 2 */
  mySensor2.start();

/* USER CODE BEGIN 3 */
mySensor.read();
mySensor2.read();
HAL_Delay(100);
```

Si descomentas el ADC::start(); el LinearSensor funciona y el DigitalSensor no. Si lo comentas el LinearSensor deja de funcionar (obviamente) y el DigitalSensor comienza a funcionar. 

Por si surge la duda, el start del DigitalSensor no hace nada, esta literalmente vacío. Por eso no lo convoque. (Lo de los starts lo arreglare en cuanto consiga que, bueno, funcionen los Sensor)

Alguna idea de que puede ser?

ricardochust, 2023-01-11:
De los ADC van:
PC0, PA3, PA4, PA5, PA0, PB0, PA6, PF11, PF12, PF13, PF14, 

No van:
PC2, PC3, PF5, PF6, PF7, PF8 (todos los del ADC3)

alejandrogonzalvo, 2023-01-13:
Solucionado

closed 2023-01-13 by alejandrogonzalvo