Hyperloop-UPV/ST-LIB #151: A la clase DigitalOutput le falta una función toggle.

PabloGonzalezSegarra, 2023-01-22:
La clase `DigitalOutput` le falta una función toggle lo cual hace engorroso hacer parpadear leds. Ahora mismo para hacer parpadear un led hay que hacer: 

```cpp
if (toggle) {
		LED_CAN.turn_on();
		toggle = false;
	}else{
		LED_CAN.turn_off()
		toggle = true;
	}
```

Además la función `set_state()` es redundante porque ya puedes poner los dos estados con las funciones `turn_on()` y `turn_off()`.

Mi propuesta:
```cpp

class DigitalOutput {
public:
	DigitalOutput() = default;
	DigitalOutput(Pin& pin);

	void turn_on();
	void turn_off();
	void toggle();
private:
	uint8_t id;
	Pin pin;
};
```

StefanCostea, 2023-01-22:
oido cocina

StefanCostea, 2023-01-22:
aunque yo dejaria set state

StefanCostea, 2023-01-22:
ya que te permite colocar el estado con una variable en runtime sin hacer comprobaciones de ifs y tal

PabloGonzalezSegarra, 2023-01-22:
vale, se puede dejar el set_state pero el toggle vendria muy bien.

ricardochust, 2023-01-22:
entonces eliminad turn_on y turn_off no? set_state ya hace lo que ambos pero mejor
o tiene alguna utilidad fuera del normal use case?

PabloGonzalezSegarra, 2023-01-23:
A ver hay que dejar los que ya estan y añadir una más que sea toggle.

PR #167 (2023-01-26): DigitalOutput Toggle
This PR adds DigitalOutput Toggle method. This PR closes #151.
closed 2023-01-26 by PabloGonzalezSegarra