Hyperloop-UPV/ST-LIB #156: EXTI sometimes triggers twice on PULL_UP

ricardochust, 2023-01-22:
the EXTI class activates an interruption each time it detects a change on a GPIO pin from OFF to ON, but sometimes it triggers twice

It seems to be a problem on the board itself (or the microprocessor), as if the voltage of the GPIO_OUT generates two slopes on a change of state instead of one, or the GPIO_IN reads two slopes on the voltage where it is only one. 

I m not sure whats the source, but we should put a protection to the EXTI class or something alike so we don t get the interrupts executed twice on a single change of state. We cannot use interruptions on critical systems if they aren t consistent. 

Code test:

```
Pin& myPin1 = PA3;
Pin& myPin2 = PC0;
DigitalOutput myOut1 = DigitalOutput(myPin1);
DigitalSensor mySensor2 = DigitalSensor(myPin2, myValue2);
uint8_t exti = ExternalInterrupt::inscribe(myPin2, [&](){step++;}).value_or(69);
```

All of it works but sometimes step is increased by 2 when myOut1 changes from OFF to ON. It doesn t seem to be an error on the wire connection as anything else works perfectly and is very consistent.

Warning: don t even bother trying to do the test by hand instead of using a DigitalOutput. When you put the wire on the pin you generate a lot of voltage noise and it can trigger up to 50 times on one state change. The exti s from the board seem to be VERY sensible to voltage changes, maybe we can configurate their thresholds or something which would solve the issue. But if thats not the case, we should implement these thresholds by software (a timeout until the interrupt can happen again would be an example)

StefanCostea, 2023-01-22:
no estoy seguro de que se pueda usar un exti y un digital input a la vez

StefanCostea, 2023-01-22:
es el caso?

PabloGonzalezSegarra, 2023-01-23:
Has medido con el osciloscopio a para ver que le esta entrando? Prueba a generar la seña con el analog discovery que la genera más limpia a ver si sigue pasando.

PabloGonzalezSegarra, 2023-01-23:
También prueba con otra nucleo.

PabloGonzalezSegarra, 2023-01-23:
También comprueba la configuración del exti a ver si lo tienes puesto en detectar flanco de subida y bajada.

closed 2023-02-01 by ricardochust