Hyperloop-UPV/ST-LIB #203: Problema con los relojes de la nucleo.

PabloGonzalezSegarra, 2023-02-07:
Hay algún problema con los relojes de la nucleo. Deben estar mal configurados porque el UART y el FDCAN no funcionan correctamente en las nucleos, mientras que funcionan correctamente en las placas. Como lo unico que cambia son los relojes al cambiar el target en el start debe de haber algo que se me escapa.

ricardochust, 2023-02-08:
como curiosidad van a 176 MHz. Hay una variable para ver la frecuencia del core clock en la HAL, llamada SystemCoreClock (es una uint32_t)

PabloGonzalezSegarra, 2023-02-16:
Acabo de provar los relojes de las nucleo y la función `HAL_RCC_GetSysClockFreq()` me devuelve que la nucleo esta funcionando a 1,718Ghz. Esto es bastante absurdo, aunque he probado el mismo código en una placa con los relojes de placas y obtengo 550Mhz. 

He revisado los relojes que hay en la ST-LIB Start y son correctos, algo raro esta pasando.

PabloGonzalezSegarra, 2023-02-16:
[image]

PabloGonzalezSegarra, 2023-02-16:
Configuración de las placas:
[image]

PabloGonzalezSegarra, 2023-02-16:
La función `HAL_RCC_GetSysClockFreq()` no devuelve las frequencia real a la que va el micro si no que la calcula a partir de los parametros que se le pasan, es decir que el micro puede perfectamente estar funcionando a una frecuencia distinta a la indiaca. Por ejemplo esto puede pasar si en la configuración se le indica que el HSE va a 25Mhz pero en realidad va a 8Mhz. Por tanto esto deja claro que lo que esta mal es la configuración de relojes de las nucleos. A pesar de esto como he comentado la he comprobado por segunda vez y no parece estar mal. Voy a volver a revisarla a ver si se me ha pasado algo.

PabloGonzalezSegarra, 2023-02-16:
Estas son las frequencias calculadas usando la configuración Nucleo:
[image]

PabloGonzalezSegarra, 2023-02-16:
Estas son las frequencias calculadas usando la configuración Board(correcta):
[image]

PabloGonzalezSegarra, 2023-02-16:
# Nucleo:
[image]
[image]

# Board:
[image]
[image]

PabloGonzalezSegarra, 2023-02-16:
# Autogenerated code for Nucleo
```cpp
void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

  /** Supply configuration update enable
  */
  HAL_PWREx_ConfigSupply(PWR_LDO_SUPPLY);

  /** Configure the main internal regulator output voltage
  */
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE0);

  while(!__HAL_PWR_GET_FLAG(PWR_FLAG_VOSRDY)) {}

  /** Macro to configure the PLL clock source
  */
  __HAL_RCC_PLL_PLLSOURCE_CONFIG(RCC_PLLSOURCE_HSE);

  /** Initializes the RCC Oscillators according to the specified parameters
  * in the RCC_OscInitTypeDef structure.
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_LSI|RCC_OSCILLATORTYPE_HSE;
  RCC_OscInitStruct.HSEState = RCC_HSE_BYPASS;
  RCC_OscInitStruct.LSIState = RCC_LSI_ON;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
  RCC_OscInitStruct.PLL.PLLM = 4;
  RCC_OscInitStruct.PLL.PLLN = 275;
  RCC_OscInitStruct.PLL.PLLP = 1;
  RCC_OscInitStruct.PLL.PLLQ = 4;
  RCC_OscInitStruct.PLL.PLLR = 2;
  RCC_OscInitStruct.PLL.PLLRGE = RCC_PLL1VCIRANGE_1;
  RCC_OscInitStruct.PLL.PLLVCOSEL = RCC_PLL1VCOWIDE;
  RCC_OscInitStruct.PLL.PLLFRACN = 0;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }

  /** Initializes the CPU, AHB and APB buses clocks
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2
                              |RCC_CLOCKTYPE_D3PCLK1|RCC_CLOCKTYPE_D1PCLK1;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.SYSCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_HCLK_DIV2;
  RCC_ClkInitStruct.APB3CLKDivider = RCC_APB3_DIV2;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_APB1_DIV2;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_APB2_DIV2;
  RCC_ClkInitStruct.APB4CLKDivider = RCC_APB4_DIV2;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_3) != HAL_OK)
  {
    Error_Handler();
  }
}

/**
  * @brief Peripherals Common Clock Configuration
  * @retval None
  */
void PeriphCommonClock_Config(void)
{
  RCC_PeriphCLKInitTypeDef PeriphClkInitStruct = {0};

  /** Initializes the peripherals clock
[... 116 lines cut ...]
		  PeriphClkInitStruct.PLL2.PLL2N = 200;
		  PeriphClkInitStruct.PLL2.PLL2P = 2;
		  PeriphClkInitStruct.PLL2.PLL2Q = 10;
		  PeriphClkInitStruct.PLL2.PLL2R = 2;
		  PeriphClkInitStruct.PLL2.PLL2RGE = RCC_PLL2VCIRANGE_0;
		  PeriphClkInitStruct.PLL2.PLL2VCOSEL = RCC_PLL2VCOMEDIUM;
		  PeriphClkInitStruct.PLL2.PLL2FRACN = 0;
		  PeriphClkInitStruct.PLL3.PLL3M = 25;
		  PeriphClkInitStruct.PLL3.PLL3N = 192;
		  PeriphClkInitStruct.PLL3.PLL3P = 2;
		  PeriphClkInitStruct.PLL3.PLL3Q = 2;
		  PeriphClkInitStruct.PLL3.PLL3R = 2;
		  PeriphClkInitStruct.PLL3.PLL3RGE = RCC_PLL3VCIRANGE_0;
		  PeriphClkInitStruct.PLL3.PLL3VCOSEL = RCC_PLL3VCOMEDIUM;
		  PeriphClkInitStruct.PLL3.PLL3FRACN = 0;
		  PeriphClkInitStruct.FdcanClockSelection = RCC_FDCANCLKSOURCE_PLL2;
		  PeriphClkInitStruct.AdcClockSelection = RCC_ADCCLKSOURCE_PLL3;

	}

	if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInitStruct) != HAL_OK) {
		ErrorHandler("The RCCEx peripheral clock did not start correctly");
	}
}
```

PabloGonzalezSegarra, 2023-02-16:
He pasado el HalInit (el de nucleo y el de board) por un checkdiff para ver si cambiaba algo y ese era el problema pero esta correcto.

PabloGonzalezSegarra, 2023-02-16:
Acabo de provar a subir código desde el template project, en vez de desde mi proyecto BLCU y ha funcionado:
[image]

La diferencia entre los dos es que BLCU esta configurado con el clock tree de placas, y por tanto generado código con esa config. El template project tiene configuración de nucleo. Por tanto hay algo que se esta autogenerando que no configuramos adecuadamente en el `STLIB::start()`. Hay que investigar de que se trata, tengo mis sospechas que es donde se especifica la frecuencia de entrada del HSE.

PabloGonzalezSegarra, 2023-02-16:
He encontrado lo que parece ser el problema. Efectivamente la frecuencia del oscilador se configuraba en el _stm32h7xx_hal_conf.h_ de la siguiente manera. Como se trata de un define autogenerado lo voy a solucionar redefiniendolo justo antes de nuestro config.

```cpp
#if !defined  (HSE_VALUE)
#define HSE_VALUE    (8000000UL) /*!< Value of the External oscillator in Hz : FPGA case fixed to 60MHZ */
#endif /* HSE_VALUE */
```

closed 2023-02-16 by PabloGonzalezSegarra