Hyperloop-UPV/ST-LIB #511: Can't compile boundaries

jorgesg82, 2025-05-07:
There was a warning that advises you that maybe you are messing up with concatenated conditions without parenthesis, and that is precisely what is going on with boundaries. In fact, the compiler is right and it is evaluating badly. Is on line 260 aprox on `Boundary.hpp` file.
The code below is the problematic one:

``` C++
Protections::FaultType check_bounds()override{
if(*src < lower_boundary || *src > upper_boundary) return Protections::FAULT;
	if(has_warning_level && *src < lower_boundary || *src > upper_boundary){
		return Protections::WARNING;
	}
	return Protections::OK;
}
```

It can be fixed this way:
``` C++
Protections::FaultType check_bounds()override{
	if(has_warning_level && (*src < lower_boundary || *src > upper_boundary)){
		return Protections::WARNING;
	}
	if(*src < lower_boundary || *src > upper_boundary) return Protections::FAULT;
	return Protections::OK;
}
```

g0nz4I0, 2025-05-07:
Oscar found more issues also with some warnings and errors saying that fault_message may be being used unitialized or something like, that, I will take a look at the compilation of the Boundaries, because I believe they are not compiled unless an instantion is created, and therefore some errors may be left undetected.

g0nz4I0, 2025-05-14:
There is already a fix for this in development, in fact it was you who fixed it in March in #499, maybe you weren't using an up to date development?

closed 2025-05-14 by g0nz4I0

jorgesg82, 2025-05-15:
Crazy. I live in the moon